QU-RoboSub/2025-general #12: Create propulsion simulation program

MFauzanAP, 2024-06-10:


MFauzanAP, 2024-09-07:
Just wanted to give an update regarding the API stuff, I've been able to get the following info for each configuration:
- Mass
- Volume
- Center of Mass
- Inertia
- Width
- Length
- Height
- Buoyancy Force
- (wip) Center of Buoyancy

As for the rest, it seems there's a way to get them alhamdulillah, just requires a bit more coding.

@khalidbelkhair could you give me a list of outputs you need from this? for e.g thruster placements, orientations, etc.

khalidbelkhair, 2024-09-07:
Sounds good!

For now the key outputs that are needed are the following for each configuration:
- Mass
- Center of Mass
- Inertia
- Thruster placements (x-y-z coordinates)
- Thruster orientation (make sure to specify whether it's front or back facing)

Something to think about as well: 
In theory we want the center of buoyancy to be directly above the center of mass. Also, we want the buoyancy force to be slightly greater than the weight. But if they are not, then the AUV will have a tendency to tilt, and sink/float. In that case, the ideal case is to play around with the masses and foams around the AUV, but of course, a control system also needs to be present to account for that. Now here is the question, should that control system rely on the variables specified earlier (CoM, CoB, etc.) to adjust the thrusters power, or should that be done from the sensor readings (closed loop feedback system). Or should it be an integrated system between the two?

MFauzanAP, 2024-09-07:
> Thruster orientation 

Could you send an example of what that might look like in code? Like is it just a directional vector?

As for your question about the control system, I think it's just the closed loop feedback system. However, the way the CoM and CoB are integrated will be through the PID parameters if I'm not mistaken.

So in the simulation, we can use the CoM and CoB to accurately simulate how the vehicle might behave in real life. Then when we simulate the control system in the virtual environment, we can tune our PID parameters properly to account for everything (mass, inertia, thruster config, CoM, CoB, etc) to ensure the vehicle can move in a stable manner.

I think this is how it'll work. If @MoharramA or someone else can provide some feedback, please do so.

khalidbelkhair, 2024-09-07:
> Could you send an example of what that might look like in code? Like is it just a directional vector?

Yes, just a directional vector. For example, (1,1,0) would be a vector facing 45 degrees in-between positive x-axis and y-axis. Or (-1,-1,0), which would be in the opposite direction.

> I think it's just the closed loop feedback system

I agree, we shouldn't have that issue theoretically speaking as we are aiming to keep it balanced (by using masses and foam). But if not, then it can just be considered as an error (just like current, etc.) and taken by the closed loop feedback system. 

We will definitely look back into this as soon as we get a better understanding inshaAllah.