QU-RoboSub/2025-general #5: Get borrowed ROV to work

MFauzanAP, 2024-06-10:


MFauzanAP, 2024-06-18:
Testing discord updates for multi-subsystem tasks.

MFauzanAP, 2024-06-19:
An update regarding the ROV, the batteries are apparently not charged and also have a weird smell coming from them (need to double check again). We need to try find a charger for these and see if they can still be used or if we need to buy new batteries.

[image]

[image]

[image]

Saad-Mohamed-Aden, 2024-06-20:
[image]
we could use this battery charger 
which I was planning anyway to put in the list 

and for the smell, I think it is better to not use these batteries, because batteries are generally dangerous, especially li ion and lipo
and smell could indicate that something is leaking so forget about using those
we could use this 
[image]

 but 
1-it does have smaller capacity which mean less time in water 
2-the size for it is bigger than the one in the picture (which means changing the design)

or we could find another battery with the same size (26650) or we could buy it from the compeny which i do not think they would be able to ship it to us
[image]

MFauzanAP, 2024-06-20:
> we could use this battery charger

Oh actually I remember seeing some of these (charger + battery) lying around in QU, maybe we could test to see if 3 of them can fit in the tube, if it can I guess we replace the current batteries w the green one.

But I don't think we should spend too much on these since it's only to get the openrov up and running. If you have these batteries already, I can bring the ROV to you in the next meeting so we can test. If not, I'll try contact someone who might have them and see if he can let us have a few.

MFauzanAP, 2024-08-30:
No need for this anymore.

closed 2024-08-30 by MFauzanAP