QU-RoboSub/2025-general #7: Update product requirements and constraints to cover all subsystems

MFauzanAP, 2024-06-10:


MFauzanAP, 2024-07-01:
@QU-RoboSub/electrical @QU-RoboSub/computer-engineering Please take a look at the [product requirements](https://qucloud.sharepoint.com/:w:/r/sites/MSDP-Newteam/Shared%20Documents/General/The-drive/General%20Files/Product%20Requirements.docx?d=w1a966b1595b145f28595995db7dc2b0a&csf=1&web=1&e=woPnpz), I've added some for your subsystems. Note that these **only contain requirements with numerical targets** so they can be put inside the House of Quality. The others will be put under the design constraints.

a-irhimeh, 2024-07-02:
I just checked it out and made some changes in the mission control, computer vision, and acoustics and communication section. I just need @MoharramA or @mombash to review the acoustics part since this is a shared subsystem.

MFauzanAP, 2024-07-02:
> I just checked it out and made some changes in the mission control, computer vision, and acoustics and communication section. I just need @MoharramA or @mombash to review the acoustics part since this is a shared subsystem.

I just took a look at the changes you've made, but actually we need to have numerical targets not qualitative ones. For example the "clear picture underwater" is hard to track and is subjective, meaning it'll be hard to measure. Either we can think of a quantitative value for it, or I think camera megapixels would be good enough tbh, and we can put "must provide clear pictures underwater" as a design constraint instead.

For the position triangulation, we can make a numeric target for it, for e.g it should calculate position with < 0.1m error or something. But I also think receiving range should be a specification that is tracked.

MFauzanAP, 2024-07-04:
I'll be closing this as complete for now, we can come back to it later.

closed 2024-07-04 by MFauzanAP