QU-RoboSub/2025-general #14: Setup automations for discord, github, and outlook

MFauzanAP, 2024-06-12:


MFauzanAP, 2024-06-14:
Discord channel for each subsystem which is open to anyone so others can get updated on what others are doing, they can mute it as well if it gets too annoying, maybe we can make it muted by default but idk if thats a thing on discord

MFauzanAP, 2024-06-14:
You can listen to events on GitHub using webhooks, see [here](https://docs.github.com/en/webhooks/webhook-events-and-payloads#issue_comment). They even include data about the issue, so theoretically you can do something like this:

```js
// data obtained whenever the issue-comment event is sent
const payload = ...

// send update to each subsystem channel
payload.issue.labels.forEach((label) => {
  ...
});
```

a-irhimeh, 2024-06-14:
Setup webhooks for each team + the general stuff

a-irhimeh, 2024-06-14:
Tst new config

MFauzanAP, 2024-06-14:
Currently this way all updates will be put in one channel, not sure if its the best idea bcs you cant mute notifs for a single subsystem

a-irhimeh, 2024-06-14:
seems like it is very limited when it comes to what can be seperated

MFauzanAP, 2024-06-14:
> seems like it is very limited when it comes to what can be seperated

If we want to separate the notifs I think we have no choice but to make a custom bot. If you've done coding with javascript before it wont be too difficult though. Check the following resources:

- [Discord.js getting started](https://discordjs.guide/)
- [Github webhooks for issue comment](**url**)

a-irhimeh, 2024-06-14:
Sure I will see what I can do

a-irhimeh, 2024-06-14:
How about this lets keep this for later and focus more on the important tasks I want to have something presentable by the end of eid week

MFauzanAP, 2024-06-14:
> How about this lets keep this for later and focus more on the important tasks I want to have something presentable by the end of eid week

Sure no worries, this is P4 anyways, but if you think theres someone thats free to do this you can assign it to them to research

MFauzanAP, 2024-06-17:
test

closed 2024-06-18 by MFauzanAP