QU-RoboSub/2025-general #21: Research Communication Tether

mombash, 2024-06-23:


MoharramA, 2024-06-23:
Please beware of components that produce too much heat such as a high power amplifier.

mombash, 2024-06-23:
### Current Subtasks:
- [ ]  Read Theory Behind VLF
- [x]  Find if RTLSDR/USRP setup can support VLF
- [ ]  Find VLF Code/GNURadio Implementation.
- [ ]  Investigate feasibility of implementations
 - [ ] N210 + RTL
 - [ ] N210 + Sound Card
 - [ ] RC Equipment
- [ ]  Ask Abdalla for proposed RF Products
- [ ]  Define possible approaches and constraints of each approach with recommended equipment
 - [ ] Constrains and Approaches
 - [ ] Equipment

### Proposed transmission reception setups:
    - USRP N210 for transmission and RTL-SDR USB dongle for reception (500 kHz)
    - USRP N210 for transmission and computer sound card for reception (~150 kHz)
    - RC car-like setup from Alibaba

a-irhimeh, 2024-06-29:
Perfect, the constraints are the same as I have envisioned exactly as well as the design