QU-RoboSub/2025-general #22: Write CPS5 Prototype Document

mombash, 2024-06-25:
## Document Description:
The document will serve as a detailed description of the following:
* What the CPS5 course is (Content: list of items, documentation; Creators: University students; etc.)
* Why we need the course (e.g. we need to gain experience)
* What we will do with the course:
  * Explain what parts will be copied over from the CPS5 course
  * Explain what parts we will design ourselves.
* What phases of prototyping will we go through

## Prototyping Phases:
* 1st Stage Prototype:
Build an ROV using a step-by-step tutorial with some slight modifications to it that will match out initial needs.
(Deadline: end of summer)

* 2nd Stage Prototype:
Transition the ROV into an AUV while maintaining most of the hardware that was used in the 1st stage prototype.
(Deadline: 1 month after fall semester starts)

* AUV final version:
Start working on the final version AUV by using all the knowledge we gained building the prototype and reusing some of the parts from it.
(Deadline: MSDP final deadline)

## Document Format:
1. **Introduction:** Introduce the CPS5 course, and why we need it
2. **Overview of prototyping phases:** Give a general overview of the prototyping phases, similar to the one give above.
3. **Details of each prototyping phase:** Give details of each prototyping phase, prioritizing the first prototyping phase. For each phase:
* Phase 1: Describe in detail what we will take from the course and what we will design ourselves.
* Phase 2: Describe in general the changes that will be made to convert the Phase 1 prototype to an AUV
* Phase 3: Explain how we will use our experience from Phases 1 and 2 to build our own prototype.

a-irhimeh, 2024-06-25:
Lets also show a comparison of the parts we needed for the previous prototype idea conpared to this or show how the same hardware can be used with minal change

a-irhimeh, 2024-06-25:
We can also add the specs of the drone in the course and what are our target specs (speed, size, weight, etc..)

MFauzanAP, 2024-06-25:
> We can also add the specs of the drone in the course and what are our target specs (speed, size, weight, etc..)

Regarding this, I've worked on a design specifications document already detailing the different specs we'd want to measure and design for. However I haven't actually set the values for these yet, so I can do this insyaAllah.

MFauzanAP, 2024-06-25:
> Lets also show a comparison of the parts we needed for the previous prototype idea conpared to this or show how the same hardware can be used with minal change

Also in this document, maybe we can show using a figure the different components that will stay from prototype to prototype.

mombash, 2024-06-25:
I've added a [link to the document on our cloud](https://qucloud-my.sharepoint.com/:w:/g/personal/mb2105947_qu_edu_qa/EWgMbXkNf41Oq-XA5U7LsacBk5tJDzSeKqN3g3MPCf9M6w?e=bYUBHR) to the [issue description](https://github.com/QU-RoboSub/2025-general/issues/22#issue-2372105953)

a-irhimeh, 2024-06-25:
I think the modified should be different here is what I have I have in mind:
1) Introduction: 
- List the old plan of the prototype and the new plan in general
- Introduce the CPS 5 course and explain it's general benefits
2) Justification:
- Justify the reason behind changing the plan by listing the financial, technical, and educational benefits for the team
3) Expectations:
- Specify the technical quantifiable goals that should be achieved by the final prototype
4) Action Plan:
- Determine an action plan with goals at the end of each phase

mombash, 2024-06-25:
Guys before working on the document, please watch [this 3 minute video](https://www.youtube.com/watch?v=w2lES-5Ynbk) in 2x speed about working with MS word styles if you haven't worked with them before, as they will help with consistent formatting and ease of navigation and indexing. I will add the link to the bottom of the description

MFauzanAP, 2024-06-25:
Should we have a style for all our documents? Just so things are consistent. I've got a very basic style going in these documents that we might be able to build off of (of course if you have a style in mind already, feel free):
- [Doc 1](https://qucloud.sharepoint.com/:w:/s/MSDP-Newteam/EVOLx2aVVIBAmJkOBoE3rAABrE6FNp1ssMCkDma3WNg2eA?e=VMdp0g)
- [Doc 2](https://qucloud.sharepoint.com/:w:/s/MSDP-Newteam/EVOLx2aVVIBAmJkOBoE3rAABrE6FNp1ssMCkDma3WNg2eA?e=VMdp0g)

mombash, 2024-06-25:
## Progress Check 1:
### Formatting:
I have made a template for the structure of [the document](https://qucloud-my.sharepoint.com/:w:/g/personal/mb2105947_qu_edu_qa/EWgMbXkNf41Oq-XA5U7LsacBk5tJDzSeKqN3g3MPCf9M6w?e=qnu6Hh) which merges both my and @a-irhimeh's proposed structures
### Content:
* I have added the decisions we made about the sections as guides to each section/subsection
* I have written a full draft for the introduction section (Section 1)
* I have written a full draft of the overview of the prototypes section (Section 2.1)
### Comments:
I have added some feedback on some of the decisions made @a-irhimeh, and notes on the drafts I wrote as [word comments](https://support.microsoft.com/en-us/office/using-modern-comments-8d3f868a-867e-4df2-8c68-bf96671641e2)
### Unimplemented:
I have yet to implement the other comments and discussions we had regarding [the budget](https://github.com/QU-RoboSub/2025-general/issues/22#issuecomment-2188368890) and [prototype comparison](https://github.com/QU-RoboSub/2025-general/issues/22#issuecomment-2188443635) into the structure of the document.
### Recommendations:
We need to divide the rest of the document between us, we should have a discussion on where the unimplemented points go, and who should work on which section (you may refer to the table of contents of [the document](https://qucloud-my.sharepoint.com/:w:/g/personal/mb2105947_qu_edu_qa/EWgMbXkNf41Oq-XA5U7LsacBk5tJDzSeKqN3g3MPCf9M6w?e=qnu6Hh) to get an idea of the sections)

mombash, 2024-06-25:
# Workflow Proposal for Writing the document:
Here are a couple ways I propose we can write the document, specifically Section 2.2:
## Proposal 1:
Each team lead will write the following details regarding the prototypes:
* **Prototype 1:**
  * What will be copied over from the CPS5 course?
  * What will be designed from scratch
  * What subsystems will be involved?
* **Prototype 2:**
  * What will be added/changed from prototype 2?
  * What new subsystems will be involved?
* **Prototype 3:**
  * **Generally:** We need to mention in general how we will use our newfound expertise and insight to decide what will change and what will stay from the CPS-based designs
  * **Specifically:** each team must mention what new systems will be added in this prototype.
### Final touches:
I will bring together all of the components written by the different teams into a cohesive Section 2.2. I will also write the general point for prototype 3.
### Styling guide:
For ease of editing and finalization, if we follow this approach, I recommend each team lead to create a new style and format all text written by them in that style, so it is clear who wrote what.

## Proposal 2:
1. @a-irhimeh will write a draft for the whole of section 2.2, trying to include all the subsystems.
2. @MFauzanAP will edit what @a-irhimeh wrote, focusing on correctly describing the mechanical deviations from the CPS5 design, and the additions (if any) for prototype 2.
3. @MoharramA will go over what those two above wrote, ensuring that they have correctly portrayed the electrical systems.

## Note:
I recommend removing [Section 3: Expectations] since I don't believe we have sufficient knowledge to make any quantitative goals or expectations regarding any prototypes beyond the specs of the CPS5 drone.

## Voting:
### @MFauzanAP 
- [ ] Proposal 1
- [ ] Proposal 2
### @a-irhimeh 
- [ ] Proposal 1
- [ ] Proposal 2
### @MoharramA 
- [ ] Proposal 1
- [ ] Proposal 2
### @mombash 
- [X] Proposal 1
- [ ] Proposal 2

MFauzanAP, 2024-06-25:
Trying out a new method/representation for writing section 2,2. Doesn't look the nicest, but imo it gives a good overview of the different phases and changes. Of course, only the mechanical sections have been mostly filled.

[image]

mombash, 2024-06-25:
i like it, not much deviation from the one i proposed, i think it will simplify the process of writing the different parts

a-irhimeh, 2024-06-27:
> ng out a new method/representation for writing section 2,2. Doesn't look the nicest, but imo it gives a good overview

that is valid

a-irhimeh, 2024-06-28:
Do we have a template for proposal 1 from @mombash?

mombash, 2024-06-28:
are you talking about the link at the top of [the description](https://github.com/QU-RoboSub/2025-general/issues/22#issue-2372105953)?

a-irhimeh, 2024-06-28:
yes it was that one I was looking for it, I will start working on it, it should be done tmrw

a-irhimeh, 2024-06-29:
Section 2.2 for CE is complete

a-irhimeh, 2024-06-29:
I think we can remove part 3 and 4 maybe? Part 3 I can see it staying but part 4 is basically just part 1 and 2

MFauzanAP, 2024-06-29:
@a-irhimeh Can you double check the subsystem distribution graph and the doc Ali made originally? I think we're getting mixed up for the design challenges subsystem.

If I'm not wrong it's supposed to be for the website and tech. documentation and stuff.

a-irhimeh, 2024-06-29:
@MFauzanAP Are you referring to the graph posted on discord? Also I need the document again I am loosing track of where the old documents are, can you send it again @MoharramA . I also dont understand what you mean by the second part

MFauzanAP, 2024-06-29:
> @MFauzanAP Are you referring to the graph posted on discord? Also I need the document again I am loosing track of where the old documents are, can you send it again @MoharramA .

Yes that graph, you can also view it on github.

>  I also dont understand what you mean by the second part

Basically if I remember correctly it was assigned to Laith who will be making the website for the team. So it's the subsystem for all extra stuff like the outreach, report, social media, etc.

Maybe we can combine it with contacting sponsors @Suhib-Alsfarini.

a-irhimeh, 2024-06-29:
ok I see what you meant

a-irhimeh, 2024-06-29:
ok let me update section 2.2 asap

a-irhimeh, 2024-06-29:
fixed

mombash, 2024-06-29:
> I think we can remove part 3 and 4 maybe? Part 3 I can see it staying but part 4 is basically just part 1 and 2

I don't think so, @a-irhimeh. Fauzan and spoke about part 3 [here](https://discord.com/channels/1246109570796097547/1246109571458666559/1255219678167699456) but part 4 remains necessary because the current report lacks mention of the following:
* How the CPS5 is cheaper than buying our own ROV, and than designing our own ROV from scratch
  * This also includes mention of how the CPS5 hardware budget can easily be accommodated for if we repurpose the second jetson budget (however, this can be left out)
* Why we learn more by copying the course than designing our own ROV or buying a readymade one

If we can mention those in another section, we can remove the 4th section without an issue.

a-irhimeh, 2024-06-29:
Yeah I kind of overlooked that part, I might try to fill it out now

a-irhimeh, 2024-06-30:
Section 4 is complete

closed 2024-06-30 by mombash