QU-RoboSub/2025-general #23: Keep track of agenda/points to update on for new member onboarding (Hazim)

MFauzanAP, 2024-06-25:


mombash, 2024-06-25:
I suggest that we can create some sort of formal onboarding that can be used to give the gist of the project to Hazim, but also be repurposed to be shown to other people like interested faculty, tiee, or even sponsors.

MFauzanAP, 2024-07-05:
Let me know what you guys think of the following agenda. General list of things to mention and show:
- What is an AUV?
- What is the robosub competition about?
- The subsystem distribution
- Quickly go through the following (let them discover more on their own)
  - Discord and how its used as the main point of communication
  - Github, what we use it for, and how it works
  - Teams and what its used for
- Our current plan with the prototypes
- Where we are currently and what were working on

Saad-Mohamed-Aden, 2024-07-05:
It is actually very good 
And this order of topics will make it easier for them to understand what is happening.
Also, we could reuse the slides that explained the robosub after adding those points to it
Then, these slides with the document would be used again for other people like interested faculty, tiee, or even sponsors.

MFauzanAP, 2024-08-30:
No need for this anymore.

closed 2024-08-30 by MFauzanAP