QU-RoboSub/2025-general #24: Create Unified Style for Documents

mombash, 2024-06-25:
Should we have a style for all our documents? Just so things are consistent. I've got a very basic style going in these documents that we might be able to build off of (of course if you have a style in mind already, feel free):
- [Doc 1](https://qucloud.sharepoint.com/:w:/s/MSDP-Newteam/EVOLx2aVVIBAmJkOBoE3rAABrE6FNp1ssMCkDma3WNg2eA?e=VMdp0g)
- [Doc 2](https://qucloud.sharepoint.com/:w:/s/MSDP-Newteam/EVOLx2aVVIBAmJkOBoE3rAABrE6FNp1ssMCkDma3WNg2eA?e=VMdp0g)

_Originally posted by @MFauzanAP in https://github.com/QU-RoboSub/2025-general/issues/22#issuecomment-2189246428_

MFauzanAP, 2024-07-01:
I think I'll be pushing this to the next sprint, we first need to have some documents for reference to figure out which styling is good and nice to work with.

But when we do complete this, it should be easy enough to apply the changes to the previous documents, as long as they're using the correct styles (ie. h1, h2, paragraph, etc.)

MFauzanAP, 2024-09-02:
Closing as we will use templates provided by TIEE

closed 2024-09-02 by MFauzanAP