QU-RoboSub/2025-general #26: Bill of Materials for the prototype

a-irhimeh, 2024-07-01:
After we have finished preparing a new action plan for the MSDP and ahead of the second team meeting we want to have a ready list that contains all the materials that will be needed during at least the first prototype stage and possibly the second stage also. We will rely on the CPS 5 drone bill of materials found (here)[https://qucloud-my.sharepoint.com/:x:/g/personal/ai2107313_qu_edu_qa/EQuvV-zzX0JJkahqutEwOW8BWxRSkcK5LVgN6uLBlwvisA?e=KGEB7S] as the base while adding some modifications to it.

a-irhimeh, 2024-07-02:
Color coded the materials using following code:
Available: Green
Needs to be purchased: Red
Needs to be researched: Yellow

Might need a way to propose multiple solutions for the same subsystem

MFauzanAP, 2024-07-03:
btw where is the document for this?

a-irhimeh, 2024-07-03:
I already linked it in the first comment but I think its not that clear so I will link it here again (https://qucloud-my.sharepoint.com/:x:/g/personal/ai2107313_qu_edu_qa/EQuvV-zzX0JJkahqutEwOW8BWxRSkcK5LVgN6uLBlwvisA?e=KGEB7S)

MFauzanAP, 2024-07-03:
Oh no I meant for our bill of materials, not the cps one.

Or are we just copying it directly? But I still feel like we should have a single excel sheet with BOMs for all prototypes.

a-irhimeh, 2024-07-03:
@mombash @MoharramA I need you guys to let me know of any of the non highlighted items are not considered part of the EE subsystems especially the tether

a-irhimeh, 2024-07-03:
> Oh no I meant for our bill of materials, not the cps one.
> 
> Or are we just copying it directly? But I still feel like we should have a single excel sheet with BOMs for all prototypes.

Tbh I think we should make a new one, I used the same one since we have nothing to change. You can make a new one and link it here maybe

MFauzanAP, 2024-07-03:
> Tbh I think we should make a new one, I used the same one since we have nothing to change. You can make a new one and link it here maybe

Alright let's wait for the others to reply, I remember @MoharramA mentioning he worked on something similar before. In any case, I think it should be put on teams for easier access.

mombash, 2024-07-04:
> @mombash @MoharramA I need you guys to let me know of any of the non highlighted items are not considered part of the EE subsystems especially the tether

i'm not sure if you still need help with this, but i don't fully get what you need from us. i remember we spoke about this over the phone but i've forgotten. also please use proper sentences next time because i'm seriously struggling to understand the sentence 💀

MFauzanAP, 2024-08-01:
Made a custom bill of materials excel sheet with columns for:
- mass
- quantity
- prices from different vendors
- minimum and actual price
- status
- percent completed
- etc

It may seem a bit complicated, up to you if you'd like to use it, but for mechanical, we will maintain our list using [this excel sheet](https://qucloud.sharepoint.com/:x:/r/sites/MSDP-Newteam/Shared%20Documents/General/The-drive/General%20Files/Bill%20of%20Materials.xlsm?d=w73517803aa8a4a58ba511da07a16eba0&csf=1&web=1&e=tRur4j) (can also be found on teams > general files)

Also note that some graphs and extra tables are automatically generated when putting data inside the excel sheet, refer to pictures 2-4

[image]

[image]

[image]

[image]

MFauzanAP, 2024-08-29:
Below is the first draft for the mech. parts list for TIEE to purchase. Note the following things:
- We don't know exact cable diameters so we'll need to buy multiple different sizes so we have the freedom to choose
- We need 2 battery hulls for swapping them out easily, one will be in the AUV, the other would be charging
- The extra items are there in case we might need them
- Thruster propeller color can be customised by BR or spray painted by us

---

General
- 12x [t200 thrusters w/ basic ESC & maroon-colored propellers](https://bluerobotics.com/store/thrusters/t100-t200-thrusters/t200-thruster-r2-rp/) (BR-100344)
- 1x [150g molykote](https://bluerobotics.com/store/watertight-enclosures/enclosure-tools-supplies/molykote/?attribute_size=150+g+tube) (BR-100558)
- 1x [vacuum pump](https://bluerobotics.com/store/watertight-enclosures/enclosure-tools-supplies/vacuum-pump-kit-r2-rp/)
- 3x [prv backfill adapter](https://bluerobotics.com/store/watertight-enclosures/prv-m10-backfill-adapter/)

Penetrators
- 3x [5-pack M10 6.5mm LC](https://bluerobotics.com/store/cables-connectors/penetrators/wlp-vp/?attribute_for-cable-diameter=WLP-M10-6.5MM-LC+%28for+6.5+mm+%C2%B1+0.3+mm+cable+diameter%29&attribute_package-quantity=1-Pack)

Electronics Hull [configurator](https://bluerobotics.com/store/watertight-enclosures/wte-vp/)
- 6"x500mm aluminum tube (BR-101411-500)
- 6" oring sealing flange (BR-102201)
- blank acrylic endcap (BR-100273-004)
- 15xM10 alum. endcap (BR-100273-001)
- prv (BR-100804)
- [**EXTRA**] 6"x400mm acrylic tube (local or from eplastics)
- [**EXTRA**] [5" acrylic dome endcap](https://bluerobotics.com/store/watertight-enclosures/non-locking-series/wte6-p-end-cap-r1/)
- [**EXTRA**] [6" enclosure clamp](https://bluerobotics.com/store/watertight-enclosures/enclosure-clamp/) (BR-102710-006)

2x Battery Hull [configurator](https://bluerobotics.com/store/watertight-enclosures/wte-vp/)
- 4"x300mm acrylic tube (BR-101052-300)
- 4" oring sealing flange (BR-100665)
- blank aluminum endcap (BR-100184)
- 5xM10 alum. endcap (BR-100875)
- prv (BR-100804)
- [**EXTRA**] [4" enclosure clamp](https://bluerobotics.com/store/watertight-enclosures/enclosure-clamp/) (BR-102710-004)

Sensor Hull [configurator](https://bluerobotics.com/store/watertight-enclosures/wte-vp/)
- 3"x150mm acrylic tube (BR-101066)
- 3" oring sealing flange (BR-100665)
- 4xM10 alum. endcap (BR-100949-004)
- 4xM10 alum. endcap (BR-100949-004)
- prv (BR-100804)
- [**EXTRA**] [3" enclosure clamp](https://bluerobotics.com/store/watertight-enclosures/enclosure-clamp/) (BR-102710-003)

a-irhimeh, 2024-08-30:
> We don't know exact cable diameters so we'll need to buy multiple different sizes so we have the freedom to choose

Lets have a meeting with sketched out cabling so we can have a better idea with how the cabling works

I also think we should discuss design and functionality again because some things will be mounted outside the AUV which we need to also keep in mind

MFauzanAP, 2024-08-30:
> > We don't know exact cable diameters so we'll need to buy multiple different sizes so we have the freedom to choose
> 
> Lets have a meeting with sketched out cabling so we can have a better idea with how the cabling works

Sounds good, the ME team will try to prepare things for that insyaAllah.

> I also think we should discuss design and functionality again because some things will be mounted outside the AUV which we need to also keep in mind

Do you mean the different hulls/enclosures or the different payloads (torpedoes, gripper, dropper, etc)? But in any case, we can have @NazarulYousef attend the meeting as well since this is his subsystem.

a-irhimeh, 2024-08-30:
> Do you mean the different hulls/enclosures or the different payloads (torpedoes, gripper, dropper, etc)? But in any case, we can have @NazarulYousef attend the meeting as well since this is his subsystem.

Yes exactly lets try to arrange a meeting as soon as possible for this. I still haven't figured out how the calendar we made works but the CE team has the exact same schedule we are free on Sunday Tuesday thursday anytime after 10 (Tuesday lab on 2) and monday Wednesday after 2

MFauzanAP, 2024-08-30:
> Yes exactly lets try to arrange a meeting as soon as possible for this. I still haven't figured out how the calendar we made works but the CE team has the exact same schedule we are free on Sunday [...]

If it needs to be done ASAP, we can have it Sunday 11-12, but lets arrange this on discord

a-irhimeh, 2024-08-30:
> If it needs to be done ASAP, we can have it Sunday 11-12, but lets arrange this on discord

it can wait a little bit but we must have it offline

MFauzanAP, 2024-08-30:
Putting this here as a future reminder.

- Weights
- Buoyancy foam
- Desiccant
- Long hex standoffs for electronics rack
- General assortment of screws and nuts
- Aluminum extrusions

MFauzanAP, 2024-08-30:
> it can wait a little bit but we must have it offline

Oh no, I meant we discuss the timing on discord, but it'll be offline

a-irhimeh, 2024-09-02:
We will stick to CPS5 course with no changes

MFauzanAP, 2024-09-02:
> We will stick to CPS5 course with no changes

I think this task should be kept open, mech and ee are not finished with their tasks yet

a-irhimeh, 2024-09-02:
but regarding the prototype nothing should be changed since we should start building it asap and finish it by the end of the week

MFauzanAP, 2024-09-02:
> but regarding the prototype nothing should be changed since we should start building it asap and finish it by the end of the week

This is also for the MSDP 1 prototype, not just for your course project prototype. For your course project, you can keep it as it is

closed 2024-09-06 by MFauzanAP