QU-RoboSub/2025-general #53: Design kill switch

MFauzanAP, 2024-09-02:


Saad-Mohamed-Aden, 2024-09-29:
### Main Design for Safe and Reliable Power Management

[image]

The system is designed to balance safety, reliability, and simplicity, while minimizing complexity. The components of the system are as follows:

**- Main Batteries**
**- Small Battery**
**- High Power Relay (HPR)**
**- BlueRobotics Switch**

**Power Flow Overview:**

**1. Main Batteries:**

The main batteries supply power to all components. They are connected to the High Power Relay (HPR), which acts as the primary control point for powering the system.

**2. High Power Relay (HPR):**
The HPR is controlled by a separate switch that is powered by the small battery. This setup ensures a clean, safe, and isolated switching mechanism.

**3. Small Battery:**
The small battery is used to power the control switch for the HPR. This may seem redundant but is crucial for ensuring waterproofing and reliability in marine environments. By isolating the control circuit with the small battery, we avoid complications related to voltage and current limitations of the main switching device.

**4. BlueRobotics Switch:**
Due to its compatibility with waterproof cases, the BlueRobotics switch is used. However, it has lower current and voltage ratings, which is why it is only responsible for controlling the small battery and the HPR, rather than the entire system directly.

**Reasoning for Design Choices:**

- Waterproofing: The use of the BlueRobotics switch ensures compatibility with waterproof cases.
- Safety & Reliability: By separating the switching circuit and power supply, the system avoids overloads and potential failures due to the lower ratings of the BlueRobotics switch.
- Simplicity: This design achieves maximum safety and reliability with minimal components and without unnecessary complexity.

### Alternative Design Using a Reed Switch

[image]

In this design, a reed switch replaces the BlueRobotics switch, offering a simpler, cost-effective solution. The key components are:

**- Boost Converter**
**- Reed Switch**
**- High Power Relay (HPR)**

**Power Flow Overview:**

**Reed Switch:**
A reed switch is used for its simplicity and ability to be activated by a magnet outside the enclosure and this reed switch would be normally open as the picture decriable. This eliminates the need for direct interaction with the internal components, maintaining waterproofing integrity.

[image]
[image]

**Boost Converter:**
The reason for including a boost converter is that the reed switches available in the market typically have low current ratings but can handle higher voltages. By boosting the voltage, the reed switch can reliably trigger the HPR without exceeding its current capacity.

**High Power Relay (HPR):**
Once the reed switch is activated by the external magnet, it connects power to the signal line of the HPR, effectively switching on the main power for the system. The boost converter ensures that the voltage is high enough for the reed switch to trigger the HPR without failure.

Saad-Mohamed-Aden, 2024-10-31:
[image]

**Components for Switching the First Prototype On and Off:**

1. **Power Supply / Controller**

2. **Resistor**  
   The resistor currently in use is 1770 ohms, chosen because it was readily available. For actual testing, however, I will need to select a resistor based on the formal calculation \( R = V/I \), where \( I \) is the current the switch can handle, and \( V \) is the signal voltage. In this case, the switch can handle up to 5A, and the signal voltage is 3V, making the smallest resistor value I could use approximately 0.6 ohms. However, it’s important to consider the wire's current limit, which is likely only around 1A, so using a higher resistance is a safer choice.

3. **Relay**  
   For this first prototype, we won’t be using the T200’s full power, so this relay should (hopefully) be able to handle the current.

In future designs, we will create a mechanism that converts the rotational movement of the switch into a linear one, allowing for a kill switch that can be pressed, which is generally a safer and more standard design.

example 

[image]

MFauzanAP, 2024-10-31:
Regarding the activation mechanism, I was thinking of either:

### 1. Pull Cord

Pull on this to activate the kill switch circuit. We can have this be far away from the actual vehicle making it safer to activate, it can even be outside the water, but idk if it might accidentally activate

### 2. Lever

Large lever gives more mechanical advantage, easier to activate, and a longer handle may be safer for us to activate since itll be further away from the thrusters

### 3. Wired Button

This is similar to the pull cord, but instead of pulling it to activate, you just press a button. For this one, we might not even need the BlueRobotics switch and can instead connect a simple red button to the kill switch circuit using the underwater cables, and the button itself will be outside the water