QU-RoboSub/2025-general #55: Complete Design Documentation for Dr. Meskin

MoharramA, 2024-09-07:
As discussed last meeting with Dr. Meskin we need to prepare a document with a minimum of 3 design choices and literature review etc fully explaining where each design idea came from and comparing them to each other. The format should be similar to the SDP final report template emailed to us by Dr. Salman.

MoharramA, 2024-09-17:
Since last meeting each of us has recieved certain critisim for our document from Dr. Meskin. Our task now is to implement those changes.

Conclusion: Everyone must have their main objective, constraints and their design ideas laid out with pros and cons for each. Try to have an idea of which design you wish to go for.

Suhib-Alsfarini, 2024-09-20:
objectives and constraints: Done
Next meeting: select your Designs

closed 2024-10-15 by MoharramA