QU-RoboSub/2025-general #58: Complete components list

MoharramA, 2024-09-29:
Each electrical subsystem complete their components list and save it in your teams file for a full integration

mombash, 2024-09-29:
should we prepare different pricepoints and evaluate alternative decisions in our list? this would be good to assess the alternatives from different perspective

or should we just give the final product? this would be faster, but will have less input

Saad-Mohamed-Aden, 2024-09-29:
[Components List.xlsx](https://github.com/user-attachments/files/17181159/Components.List.xlsx)
from my side i did a components list but this just the most important parts like cables, converters etc..
my main parts like batteries i can choose theme later on while on the process but as a start i would by these