QU-RoboSub/2025-general #71: Make gate look like competition gate

MFauzanAP, 2025-01-19:


MFauzanAP, 2025-01-19:
@NazarulYousef Please add any relevant subtasks and deadlines

NazarulYousef, 2025-01-19:
Task Description:
The approved pre-qualification course consists of:
(1) horizontal Gate
(2) vertical Marker

Gate Specifications:
- Length: 6.6 ft (2 m).
- Position: 3.3 ft (1 m) below the surface of the water.
- Diameter and color: Any, as chosen by the team.

Gate Anchoring:
- Can be anchored to the floor using any method (e.g., PVC elbows to create two vertical legs).
- Lines can be attached to each end to moor the Gate to the floor.

Vertical Marker Specifications:
- Diameter and color: Any, as chosen by the team.
- Position: 33 ft (10 m) beyond the Gate.
- Must touch the floor and extend above the water's surface.

[image]

NazarulYousef, 2025-01-19:
Task Deadline:
- Horizontal Gate **26/01/2025**
- Vertical Marker **28/01/2025**